PRBonn/rangenet_lib #4: only support  building with CUDA7.5?

KevinYuk, 2019-11-07:
```

[build] Found '2' packages in 0.0 seconds.
[build] Package table is up to date.
Starting  >>> catkin
Finished  <<< catkin                      [ 0.1 seconds ]
Starting  >>> rangenet_lib
____________________________________________________________________________________________________________________________________________________________________________________
Errors     << rangenet_lib:cmake /home/2T/patent_exper/3DFGFA/rangenet_lib/catkin_ws/logs/rangenet_lib/build.cmake.004.log
Build type: Release
CMake Error at /usr/local/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CUDA: Found unsuitable version "10.0", but required is exact
  version "7.5" (found /usr/local/cuda)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:375 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.9/Modules/FindCUDA.cmake:1020 (find_package_handle_standard_args)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:45 (find_package)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:250 (find_host_package)
  CMakeLists.txt:25 (find_package)
```

Chen-Xieyuanli, 2019-11-07:
Thanks for using our code.

We don't specify the version of cuda in our code. It may be caused by the wrong version of TensorRT.

Please try to find a proper one, which matches your cuda version, here: https://developer.nvidia.com/nvidia-tensorrt-download.

KevinYuk, 2019-11-07:
Hi @Chen-Xieyuanli :
Thanks for your reply.

I have tried TensorRT 5.0 and 6.0. Both don't work.
Below is what I have tried:
1. cuda7.5 + tensorRT5.0
`error kVERBOSE is not a member of nvinfer1::ILogger::Severity`
2. cuda10  + tensorRT6.0
```
Errors     << rangenet_lib:cmake /home/2T/patent_exper/3DFGFA/rangenet_lib/catkin_ws/logs/rangenet_lib/build.cmake.000.log
Build type: Release
CMake Error at /usr/local/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CUDA: Found unsuitable version "10.0", but required is exact
  version "7.5" (found /usr/local/cuda)
Call Stack (most recent call first):

```

jbehley, 2019-11-07:
I think that the particular problem is caused by an old version of OpenCV, as the stack trace suggests,

```
  /usr/local/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:375 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.9/Modules/FindCUDA.cmake:1020 (find_package_handle_standard_args)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:45 (find_package)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:250 (find_host_package)
```
I suspect that you are still using a "stock" version from Ubuntu of OpenCV 2.4.x, which might be compiled with CUDA 7.5?

I removed OpenCV as a required dependency from the CMakefile, since it is only needed for the example. Thus, it should now compile with CUDA 10 and TensorRT. If you want to have also the example visualization you need to install OpenCV 3.0+ and this should be then also be compiled with an proper version or at least should not require in its CMakefile an older version of CUDA.

Does this solve your problem?

jbehley, 2019-11-29:
@KevinYuk I close the issue for now, since it seems to be not a problem anymore.

closed 2019-11-29 by jbehley