PRBonn/rangenet_lib #6: Not getting expected results

Ewaolx, 2019-11-12:
Hi,

I just tried inferring on one of the sample data from 64 channel lidar. It looks like it is not getting segmented properly. So is it have to do with setting some parameters and training it, as I am using the pretrained model. Below is the result. Left is the input.

[image]

Thanks.

Chen-Xieyuanli, 2019-11-21:
Sorry for the late reply.

Several reasons may cause this problem.

1. Our rangenet_lib and the provided pretrained model can now only work well with KITTI LiDAR data. It may provide wrong predictions when you test with other LiDAR data.
2. The runtime tensorrt model will be generated depending on your hardware. It may not work with other versions of the graphic cards or drivers. Could you please provide more information about your hardware setups. We may also have a test later.

Since this issue is duplicate with another similar issue [here](https://github.com/PRBonn/semantic_suma/issues/6), I will close this one.

closed 2019-11-21 by Chen-Xieyuanli

Ewaolx, 2019-12-08:
Sorry for late reply and reopening this issue. 

Below is my hardware setup.
- GeForce GTX 1050
- Cuda: 10.1, V10.1.243
- Cudnn: 7
- TensorRT: 6.0.1

Just wanted to know, if this setup is compatible or should I switch to other setup.

Thanks!