PRBonn/rangenet_lib #7: Exporting PyTorch models to ONNX

ongdexter, 2019-11-22:
Hi, 

Thanks for sharing your work. I am interested in deploying the other RangeNet++ models on TensorRT as well. I've tried exporting them to onnx by tracing but faced issues parsing them on TensorRT. Will you be able to share your scripts or modifications for exporting the models from PyTorch to ONNX?

Many thanks!

Chen-Xieyuanli, 2019-11-26:
Sorry for the late reply.

This repo is only an example library for our semantic LiDAR SLAM, SuMa++. We will later release a full C++ develop repo for RangeNet++, which will include the scripts for exporting the models in different formats.

ongdexter, 2019-11-27:
That's understandable, thanks for getting back.

closed 2019-11-27 by ongdexter

aditbhrgv, 2020-01-10:
Hello,

Are the scripts for converting the Pytorch trained models available?
I am trying to train another model from scratch and would like to see its performance with TensorRT as well.
Thanks in advance!

BR

Chen-Xieyuanli, 2020-01-20:
Hey @aditbhrgv, sorry for the late reply. here is my hack, which is not a well-programmed, but you could take as an example.
[make_onnx.tar.gz](https://github.com/PRBonn/rangenet_lib/files/4086105/make_onnx.tar.gz)

The final version for deploying will be released soon in [Lidar-bonnetal](https://github.com/PRBonn/lidar-bonnetal)

akouri-dd, 2020-05-08:
Hi @Chen-Xieyuanli - is there a branch open for this the relates to darknet for rangenet?

Chen-Xieyuanli, 2020-05-09:
Hi @akouri-dd, unfortunately I have no plan to create a new branch in Lidar-bonnetal currently to generate the ONNX model.

akouri-dd, 2020-05-19:
Thanks. I opened up a fork that contains the modified version of the script to generate the ONNX file. It is pretty rough. You can see that here: https://github.com/andrewkouri/lidar-bonnetal/blob/master/train/tasks/semantic/create_onnx.py

One problem that I am running into now is that the models do not run with the inference code in this repo because the ir_version of the model is too high (0.0.6) vs (0.0.3).

So I am going to try using TensorRT 7.0 instead of 5.0.

The README in this repo says that it will only work with version 5.0. Is that true? What is the issue with a version greater than 5.0?

Chen-Xieyuanli, 2020-05-20:
Hey @akouri-dd, thank you for the feedback.  So far, we only tried with 5.0 and we have received some reports from other users that our code doesn't work well with higher TensorRT versions. More information you could find in #9.

I'm not sure whether it works or not with 7.0. I will appreciate it a lot if you could provide your result.

akouri-dd, 2020-05-20:
Ok, I got it working with TensorRT 7.1 and exporting the model with the latest version of PyTorch. I will try to neaten up my code and contribute back when I have a chance.  It wasn't too tricky to do the TensorRT upgrade.

Chen-Xieyuanli, 2020-05-21:
I'm very happy to hear that TensorRT 7.1 also works for you and looking forward to your pull request :-).