PRBonn/rangenet_lib #8: What training dataset do you use?

KevinYuk, 2019-11-26:
Hi guys,

Thanks for your work.

Could you please tell what training dataset do you use?

jbehley, 2019-11-26:
Hi,

we used the SemanticKITTI training data (see also http://www.semantic-kitti.org/) for training. More specifically, we used sequence 00-10, except sequence 08, for training. Sequence 08 was used as validation set.

Please see also https://github.com/PRBonn/lidar-bonnetal for more information on the pre-trained models and the code used to train the network.

closed 2019-11-29 by jbehley

G12311231, 2024-07-02:
> 你好，
> 
> 我们使用 SemanticKITTI 训练数据（另请参阅http://www.semantic-kitti.org/）进行训练。更具体地说，我们使用序列 00-10（序列 08 除外）进行训练。序列 08 用作验证集。
> 
> 有关预训练模型和用于训练网络的代码的更多信息，请参阅[https://github.com/PRBonn/lidar-bonnetal 。](https://github.com/PRBonn/lidar-bonnetal)

Hi,
If I use the same sensor vlp-64 to collect data on my campus, I wonder if this dataset will be able to use the parameters of the pre-trained model?