PRBonn/rangenet_lib #12: Compiling issue

zzhh00, 2019-12-09:
**TensorRT version**: TensorRT 5.0.2.6 GA for Ubuntu 1604 and CUDA 9.0 DEB local repo packages
**Ubuntu version**: Ubuntu 16.04
**cuda version**: cuda 9.0

When I compile the project,error occurs as below:
 ----
netTensorRT.cpp:545:47:error: ‘kVERBOSE’ is not a member of ‘nvinfer1::ILogger::Severity’
                              static_cast<int>(ILogger::Severity::kVERBOSE))) 

Then I find that there are four members of "ILogger::Severity",however "kVERBOSE" is not in them.The requirement written in README for the tensorrt verion is 5,maybe it should be 6?

jbehley, 2019-12-11:
Please update your version at least to 5.1.0. There the option `kVERBOSE` has been added.

See also https://docs.nvidia.com/deeplearning/sdk/tensorrt-archived/tensorrt-510/tensorrt-api/c_api/classnvinfer1_1_1_i_logger.html#a1a8312e2a57dadab516e8cc251ca70dd

versus

your version:

https://docs.nvidia.com/deeplearning/sdk/tensorrt-archived/tensorrt-502/tensorrt-api/c_api/classnvinfer1_1_1_i_logger.html#a1a8312e2a57dadab516e8cc251ca70dd

Chen-Xieyuanli, 2020-01-09:
Hi @zzhh00,

Is there any update on this issue?

Chen-Xieyuanli, 2020-01-20:
Since there is no update for this issue, I will then close it.

If there is any need, feel free to ask me to reopen this issue.

closed 2020-01-20 by Chen-Xieyuanli