PRBonn/rangenet_lib #13: (Compiling Issue) Cannot find TensorRT

Pamphlett, 2019-12-12:
Hello,

Thx for your code. However I ran into trouble while executing the **catkin build**  step.
Seems it can not find Tensorrt, but I can successfully import tensorrt in Python.
My setups are as follows: GeForce GTX 1050 Ti, Cuda 10.0.130, Cudnn 7.5.0, TensorRT 5.1.5.0
Could you please help me with this?

**Workspace configuration appears valid.**

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin                                                           
Finished  <<< catkin                      [ 4.6 seconds ]                      
Starting  >>> rangenet_lib                                                     
_______________________________________________________________________________
Errors     << rangenet_lib:cmake /home/pam/catkin_ws/logs/rangenet_lib/build.cmake.000.log
Build type: Release
YAML Libs: yaml-cpp
YAML Headers: /usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../include
Boost Libs: /usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost Headers: /usr/include

TensorRT NOT Available
TensorRT NOT FOUND

CMake Error at /home/pam/catkin_ws/src/rangenet_lib/CMakeLists.txt:59 (message):
  TensorRT must be installed

    TENSORRT_FOUND OFF

cd /home/pam/catkin_ws/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/bin/cmake /home/pam/catkin_ws/src/rangenet_lib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pam/catkin_ws/devel/.private/rangenet_lib -DCMAKE_INSTALL_PREFIX=/home/pam/catkin_ws/install; cd -
...............................................................................
Failed     << rangenet_lib:cmake          [ Exited with code 1 ]               
Failed    <<< rangenet_lib                [ 1.4 seconds ]

Chen-Xieyuanli, 2019-12-12:
Hi @Pamphlett,

Could you please provide the information about executing $ dpkg -l | grep TensorRT

Pamphlett, 2019-12-12:
@Chen-Xieyuanli Nothing shows after executing.

Chen-Xieyuanli, 2019-12-12:
@Pamphlett TensorRT may not be installed properly. You could reinstall TensorRT following the [Installation Guide](https://docs.nvidia.com/deeplearning/sdk/tensorrt-install-guide/index.html#abstract)

Pamphlett, 2019-12-12:
@Chen-Xieyuanli 
I noticed just now that none of the packages nor libraries (Cuda, NVINFER, NVINFERPLUGIN... as is shown in **tensorrt-config.cmake**) required for this project could be found.

LongruiDong, 2019-12-19:
> @Chen-Xieyuanli
> I noticed just now that none of the packages nor libraries (Cuda, NVINFER, NVINFERPLUGIN... as is shown in **tensorrt-config.cmake**) required for this project could be found.

Hi, I installed TensorRT using Tar File, so 
`$ dpkg -l | grep TensorRT` may not suitable to check whether TensorRT is installed...
and I also meet same issue with you.
Have you fsolved it ? How to let cmake find tensorrt-related stuffs...

Thanks~

LongruiDong, 2019-12-19:
I solved it by adding `set(CMAKE_PREFIX_PATH "/.../TensorRT-5.1.5.0/lib")` to CMakeLists.txt

LongruiDong, 2019-12-19:
> > @Chen-Xieyuanli
> > I noticed just now that none of the packages nor libraries (Cuda, NVINFER, NVINFERPLUGIN... as is shown in **tensorrt-config.cmake**) required for this project could be found.
> 
> Hi, I installed TensorRT using Tar File, so
> `$ dpkg -l | grep TensorRT` may not suitable to check whether TensorRT is installed...
> and I also meet same issue with you.
> Have you solved it ? How to let cmake find tensorrt-related stuffs...
> 
> Thanks~

I solved it by adding set(CMAKE_PREFIX_PATH "/.../TensorRT-5.1.5.0/lib") to CMakeLists.txt

closed 2019-12-19 by Pamphlett

leoriom, 2020-10-11:
> @Chen-Xieyuanli
> I noticed just now that none of the packages nor libraries (Cuda, NVINFER, NVINFERPLUGIN... as is shown in **tensorrt-config.cmake**) required for this project could be found.
 I also meet same issue with you.
Have you solved it ?

SlamCabbage, 2022-03-25:
Thank you for your outstanding contribution, I also encountered the problem of not finding TensorRT.

-- ==> add_subdirectory(rangenet_lib)
Build type: Release
YAML Libs: yaml-cpp
YAML Headers: /usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../../include
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: program_options filesystem 
Boost Libs: Boost::program_options;Boost::filesystem
Boost Headers: /usr/include

TensorRT NOT Available
TensorRT NOT FOUND

CMake Error at rangenet_lib/CMakeLists.txt:59 (message):
  TensorRT must be installed

    TENSORRT_FOUND OFF

-- Configuring incomplete, errors occurred!
See also "/home/llz/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/llz/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

This is my output by $ dpkg -l | grep TensorRT: 
ii  libnvinfer-bin                                              7.2.1-1+cuda11.1                    amd64        TensorRT binaries
ii  libnvinfer-dev                                              7.2.1-1+cuda11.1                    amd64        TensorRT development libraries and headers
ii  libnvinfer-doc                                              7.2.1-1+cuda11.1                    all          TensorRT documentation
ii  libnvinfer-plugin-dev                                       7.2.1-1+cuda11.1                    amd64        TensorRT plugin libraries
ii  libnvinfer-plugin7                                          7.2.1-1+cuda11.1                    amd64        TensorRT plugin libraries
ii  libnvinfer-samples                                          7.2.1-1+cuda11.1                    all          TensorRT samples
ii  libnvinfer7                                                 7.2.1-1+cuda11.1                    amd64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                                        7.2.1-1+cuda11.1                    amd64        TensorRT ONNX libraries
ii  libnvonnxparsers7                                           7.2.1-1+cuda11.1                    amd64        TensorRT ONNX libraries
ii  libnvparsers-dev                                            7.2.1-1+cuda11.1                    amd64        TensorRT parsers libraries
ii  libnvparsers7                                               7.2.1-1+cuda11.1                    amd64        TensorRT parsers libraries
ii  tensorrt                                                    7.2.1.6-1+cuda11.1                  amd64        Meta package of TensorRT

arjunskumar, 2023-07-18:
@SlamCabbage have you managed to solve it?