PRBonn/rangenet_lib #14: fatal error: NvOnnxParser.h: No such file or directory

LongruiDong, 2019-12-19:
Hi,
when building the project, I met:
`-------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                                                   
[build] Package table is up to date.                                                                                                         
Starting  >>> catkin                                                                                                                         
Finished  <<< catkin                      [ 21.9 seconds ]                                                                                   
Starting  >>> rangenet_lib                                                                                                                   
_____________________________________________________________________________________________________________________________________________
Warnings   << rangenet_lib:cmake /home/dlr/Project/suma++/logs/rangenet_lib/build.cmake.005.log                                              
Build type: Release
YAML Libs: yaml-cpp
YAML Headers: /usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../include
Boost Libs: /usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost Headers: /usr/include

TensorRT available!
CUDA Libs: /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so
CUDA Headers: /usr/local/cuda/include
NVINFER: /home/dlr/Downloads/TensorRT-5.1.5.0/lib/libnvinfer.so
NVINFERPLUGIN: /home/dlr/Downloads/TensorRT-5.1.5.0/lib/libnvinfer_plugin.so
NVPARSERS: /home/dlr/Downloads/TensorRT-5.1.5.0/lib/libnvparsers.so
NVONNXPARSER: /home/dlr/Downloads/TensorRT-5.1.5.0/lib/libnvonnxparser.so
NVONNXPARSERRUNTIME: /home/dlr/Downloads/TensorRT-5.1.5.0/lib/libnvonnxparser_runtime.so
All togheter now (libs): /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so;nvinfer;nvinfer_plugin;nvparsers;nvonnxparser;nvonnxparser_runtime
All togheter now (inc): /usr/local/cuda/include
TensorRT SUCCESS!

Building TensorRT
Building example...
OpenCV Libs: opencv_core;opencv_viz
OpenCV Headers: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv

cd /home/dlr/Project/suma++/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/bin/cmake /home/dlr/Project/suma++/src/rangenet_lib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dlr/Project/suma++/devel/.private/rangenet_lib -DCMAKE_INSTALL_PREFIX=/home/dlr/Project/suma++/install; cd -
.............................................................................................................................................
_____________________________________________________________________________________________________________________________________________
Errors     << rangenet_lib:make /home/dlr/Project/suma++/logs/rangenet_lib/build.make.003.log                                                
In file included from /home/dlr/Project/suma++/src/rangenet_lib/src/netTensorRT.cpp:7:0:
/home/dlr/Project/suma++/src/rangenet_lib/include/netTensorRT.hpp:10:26: fatal error: NvOnnxParser.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/rangenet_lib.dir/src/netTensorRT.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/dlr/Project/suma++/src/rangenet_lib/include/selector.hpp:16:0,
                 from /home/dlr/Project/suma++/src/rangenet_lib/src/selector.cpp:8:
/home/dlr/Project/suma++/src/rangenet_lib/include/netTensorRT.hpp:10:26: fatal error: NvOnnxParser.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/rangenet_lib.dir/src/selector.cpp.o] Error 1
make[1]: *** [CMakeFiles/rangenet_lib.dir/all] Error 2
make: *** [all] Error 2
cd /home/dlr/Project/suma++/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.............................................................................................................................................
Failed     << rangenet_lib:make           [ Exited with code 2 ]                                                                             
Failed    <<< rangenet_lib                [ 11.6 seconds ]                                                                                   
[build] Summary: 1 of 2 packages succeeded.                                                                                                  
[build]   Ignored:   None.                                                                                                                   
[build]   Warnings:  1 packages succeeded with warnings.                                                                                     
[build]   Abandoned: None.                                                                                                                   
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 33.9 seconds total.                                            `

I find  NvOnnxParser.h in 
`TensorRT-5.1.5.0/include`, so How can i solve it?

Thanks!!

Pamphlett, 2019-12-19:
@LongruiDong 
I managed to solve it by just copy-pasting the related .h files into the current project.
I hope it helps.

LongruiDong, 2019-12-19:
> @LongruiDong
> I managed to solve it by just copy-pasting the related .h files into the current project.
> I hope it helps.

yeah! Thanks it works~

closed 2019-12-19 by LongruiDong