PRBonn/rangenet_lib #18: catkin build error

wolf943134497, 2020-01-04:
Hi @Chen-Xieyuanli   Thanks for your great works!
I install TensorRT by tar file. the error as fellows:

Build type: Release
YAML Libs: yaml-cpp
YAML Headers: /usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../include
Boost Libs: /usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost Headers: /usr/include

TensorRT available!
CUDA Libs: /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so
CUDA Headers: /usr/local/cuda/include
NVINFER: /home/TensorRT-5.1.5.0/lib/libnvinfer.so
NVINFERPLUGIN: /home/TensorRT-5.1.5.0/lib/libnvinfer_plugin.so
NVPARSERS: /home/TensorRT-5.1.5.0/lib/libnvparsers.so
NVONNXPARSER: /home/TensorRT-5.1.5.0/lib/libnvonnxparser.so
NVONNXPARSERRUNTIME: /home/TensorRT-5.1.5.0/lib/libnvonnxparser_runtime.so
All togheter now (libs): /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so;nvinfer;nvinfer_plugin;nvparsers;nvonnxparser;nvonnxparser_runtime
All togheter now (inc): /usr/local/cuda/include
TensorRT SUCCESS!

Building TensorRT
Building example...
OpenCV Libs: opencv_core;opencv_viz
OpenCV Headers: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv

cd /home/catkin_ws/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/local/bin/cmake /home/catkin_ws/src/rangenet_lib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/catkin_ws/install -DCMAKE_CXX_STANDARD=14 -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
_______________________________________________________________________________
Errors     << rangenet_lib:make /home/catkin_ws/logs/rangenet_lib/build.make.000.log
/usr/bin/ld: can not find  -lnvinfer
/usr/bin/ld: can not find -lnvinfer_plugin
/usr/bin/ld: can not find -lnvparsers
/usr/bin/ld: can not find -lnvonnxparser
/usr/bin/ld: can not find -lnvonnxparser_runtime

seems already find the so file in tensorrt installation path,but can not load

any suggestions? Thanks!

closed 2020-01-05 by wolf943134497

Chen-Xieyuanli, 2020-01-06:
Hey @wolf943134497, Sorry for the late reply.

I'm so happy that you've solved this problem. That would be very nice if you could also provide your solutions here.

Thanks!

wolf943134497, 2020-01-06:
> Hey @wolf943134497, Sorry for the late reply.
> 
> I'm so happy that you've solved this problem. That would be very nice if you could also provide your solutions here.
> 
> Thanks!

Hi @Chen-Xieyuanli ，Thanks for sharing your great works!
I solved the problem by copy the so files into /usr/lib.

Chen-Xieyuanli, 2020-01-06:
@wolf943134497 Thank you for your feedback!