PRBonn/rangenet_lib #19: Cannot build successfully !

aditbhrgv, 2020-01-07:
Hello,
I cannot compile the code successfully as I get below errors in **BOLD**. 
Can anyone help ? Thanks 

 
./catkin_ws$ catkin build rangenet_lib
---------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/adit/catkin_ws
---------------------------------------------------------
Build Space:        [exists] /home/adit/catkin_ws/build
Devel Space:        [exists] /home/adit/catkin_ws/devel
Install Space:      [unused] /home/adit/catkin_ws/install
Log Space:          [exists] /home/adit/catkin_ws/logs
Source Space:       [exists] /home/adit/catkin_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------

---------------------------------------------------------
WARNING: Your workspace is not extending any other
result space, but it is set to use a `linked` devel
space layout. This requires the `catkin` CMake package
in your source space in order to be built.
---------------------------------------------------------

[build] Found '2' packages in 0.0 seconds.                                                                                                                                                                                                                                 
[build] Package table is up to date.                                                                                                                                                                                                                                       
Starting  >>> catkin                                                                                                                                                                                                                                                       
Finished  <<< catkin                      [ 0.2 seconds ]                                                                                                                                                                                                                  
Starting  >>> rangenet_lib                                                                                                                                                                                                                                                 
Finished  <<< rangenet_lib                [ 0.3 seconds ]                                                                                                                                                                                                                  
[build] Summary: All 2 packages succeeded!                                                                                                                                                                                                                                 
[build]   Ignored:   None.                                                                                                                                                                                                                                                 
[build]   Warnings:  None.                                                                                                                                                                                                                                                 
[build]   Abandoned: None.                                                                                                                                                                                                                                                 
[build]   Failed:    None.                                                                                                                                                                                                                                                 
[build] Runtime: 0.5 seconds total.                                                                                                                                                                                                                                        
**Exception ignored in: <bound method BaseEventLoop.__del__ of <_UnixSelectorEventLoop running=False closed=True debug=False>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/trollius/base_events.py", line 395, in __del__
  File "/usr/local/lib/python3.6/dist-packages/trollius/unix_events.py", line 65, in close
  File "/usr/local/lib/python3.6/dist-packages/trollius/unix_events.py", line 166, in remove_signal_handler
  File "/usr/lib/python3.6/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object**

aditbhrgv, 2020-01-07:
I resolved the above issue. It was python3 issue. Got rid of it by using Python 2.7 however, there is no error now in "catkin build rangenet_lib" but no dlevel/lib/rangenet_lib/ folder is created.

adit@gibson2:~/catkin_ws$ catkin build rangenet_lib
---------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/adit/catkin_ws
---------------------------------------------------------
Build Space:        [exists] /home/adit/catkin_ws/build
Devel Space:        [exists] /home/adit/catkin_ws/devel
Install Space:      [unused] /home/adit/catkin_ws/install
Log Space:         [missing] /home/adit/catkin_ws/logs
Source Space:       [exists] /home/adit/catkin_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------
NOTE: Forcing CMake to run for each package.
---------------------------------------------------------

---------------------------------------------------------
WARNING: Your workspace is not extending any other
result space, but it is set to use a `linked` devel
space layout. This requires the `catkin` CMake package
in your source space in order to be built.
---------------------------------------------------------

[build] Found '2' packages in 0.0 seconds.                                                                                                                                                                        
[build] Updating package table.                                                                                                                                                                                   
Starting  >>> catkin                                                                                                                                                                                              
Finished  <<< catkin                      [ 4.1 seconds ]                                                                                                                                                         
Starting  >>> rangenet_lib                                                                                                                                                                                        
__________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << rangenet_lib:cmake /home/adit/catkin_ws/logs/rangenet_lib/build.cmake.000.log                                                                                                                       
Build type: Release
YAML Libs: yaml-cpp
YAML Headers: /usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../include
Boost Libs: /usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost Headers: /usr/include

TensorRT available!
CUDA Libs: /usr/local/cuda-10.0/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so
CUDA Headers: /usr/local/cuda-10.0/include
NVINFER: /usr/lib/x86_64-linux-gnu/libnvinfer.so
NVINFERPLUGIN: /usr/lib/x86_64-linux-gnu/libnvinfer_plugin.so
NVPARSERS: /usr/lib/x86_64-linux-gnu/libnvparsers.so
NVONNXPARSER: /usr/local/lib/libnvonnxparser.so
NVONNXPARSERRUNTIME: /usr/local/lib/libnvonnxparser_runtime.so
All togheter now (libs): /usr/local/cuda-10.0/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/x86_64-linux-gnu/librt.so;nvinfer;nvinfer_plugin;nvparsers;nvonnxparser;nvonnxparser_runtime
All togheter now (inc): /usr/local/cuda-10.0/include
TensorRT SUCCESS!

Building TensorRT
CMake Warning at /home/adit/catkin_ws/src/rangenet_lib/CMakeLists.txt:156 (find_package):
  Found package configuration file:

    /usr/local/share/OpenCV/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

cd /home/adit/catkin_ws/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/local/bin/cmake /home/adit/catkin_ws/src/rangenet_lib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/adit/catkin_ws/devel/.private/rangenet_lib -DCMAKE_INSTALL_PREFIX=/home/adit/catkin_ws/install; cd -
..................................................................................................................................................................................................................
Finished  <<< rangenet_lib                [ 9.9 seconds ]                                                                                                                                                         
[build] Summary: All 2 packages succeeded!                                                                                                                                                                        
[... cut ...]
[build]   Failed:    None.                                                                                                                                                                                        
[build] Runtime: 14.1 seconds total.                                                                                                                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

closed 2020-01-08 by aditbhrgv

BraedenCullenCodes, 2020-03-22:
how did you solve this error?

aditbhrgv, 2020-03-23:
I think I changed my Python from 3.6 to 2.7 ! Please check if it works for you !