PRBonn/rangenet_lib #20: unable to build rangenet_lib using catkin

snj00070, 2020-01-08:
Hi, I am having an issue with building Rangenet.
I have followed every step in order.
what's the issue here?

--------------------------------------------------------------------------------------------------------------------
harry@pc:/$ mkdir -p ~/catkin_ws/src
harry@pc:/$ cd ~/catkin_ws/src
harry@pc:~/catkin_ws/src$ git clone https://github.com/ros/catkin.git 
Cloning into 'catkin'...
remote: Enumerating objects: 12545, done.
remote: Total 12545 (delta 0), reused 0 (delta 0), pack-reused 12545
Receiving objects: 100% (12545/12545), 3.48 MiB | 3.30 MiB/s, done.
Resolving deltas: 100% (6786/6786), done.
Checking connectivity... done.
harry@pc:~/catkin_ws/src$ git clone https://github.com/PRBonn/rangenet_lib.git
Cloning into 'rangenet_lib'...
remote: Enumerating objects: 68, done.
remote: Counting objects: 100% (68/68), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 68 (delta 30), reused 45 (delta 15), pack-reused 0
Unpacking objects: 100% (68/68), done.
Checking connectivity... done.
harry@pc:~/catkin_ws/src$ cd .. && catkin init
Catkin workspace `/home/harry` is already initialized. No action taken.
------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/harry
------------------------------------------------
Build Space:       [missing] /home/harry/build
Devel Space:       [missing] /home/harry/devel
Install Space:      [unused] /home/harry/install
Log Space:         [missing] /home/harry/logs
Source Space:      [missing] /home/harry/src
DESTDIR:            [unused] None
------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------

------------------------------------------------
WARNING: Your workspace is not extending any
other result space, but it is set to use a
`linked` devel space layout. This requires the
`catkin` CMake package in your source space in
order to be built.
------------------------------------------------
WARNING: Source space `/home/harry/src` does
not yet exist.
------------------------------------------------

harry@pc:~/catkin_ws$ sudo catkin build rangenet_lib
[build] Error: Unable to find source space `/home/harry/src`

Chen-Xieyuanli, 2020-01-09:
Hey @snj00070,

Thanks for using our code. As the error msg says, there is something wrong with the source space configuration.

You could try to specify the source space explicitly by $ catkin config --source-space /your/source/space, and recompile the package.

Or it may be caused by the .catkin_tools from a previous wrong installation, you could remove it by $rm -rf ~/.catkin_tools, and recompile the package.

Chen-Xieyuanli, 2020-01-20:
Hey @snj00070, is there any update about this issue?

snj00070, 2020-01-21:
Hi, thank you for your kind recommendation. I was busy on working on something else at the moment. I will give you an update when I get back on it. thanks again.

Chen-Xieyuanli, 2020-03-05:
Since there is no further update, I will close this issue. Feel free to ask me to reopen it.

closed 2020-03-05 by Chen-Xieyuanli