PRBonn/rangenet_lib #22: Unable to load example

Spartacus782, 2020-01-31:
Hi,

Thanks for sharing your code, really excited to see it working, but I'm having the following issue...

./devel/lib/rangenet_lib/infer -p /home/stucky/catkin_ws2/src/darknet53 -s /home/stucky/catkin_ws2/src/rangenet_lib/example/000000.bin 
================================================================================
scan: /home/stucky/catkin_ws2/src/rangenet_lib/example/000000.bin
path: /home/stucky/catkin_ws2/src/darknet53/
verbose: 0
================================================================================
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /home/stucky/catkin_ws2/src/darknet53//model.trt
Could not deserialize TensorRT engine. 
Generating from sratch... This may take a while...
Trying to generate trt engine from : /home/stucky/catkin_ws2/src/darknet53//model.onnx
Segmentation fault (core dumped)

Any help would be appreciated.

Thanks!

Chen-Xieyuanli, 2020-02-01:
Hey @Spartacus782,

Thank you for using our code.

This issue looks quite similar to #5. Could you please have a look there?

Spartacus782, 2020-02-03:
Thanks for the quick response

In my error, it does not try the modeling.  The core segmentation fault happens immediately and the program exits.  Program exits in less than a second.  Would a stack trace help?  I will include it below

> $ valgrind -v ./devel/lib/rangenet_lib/infer -p /home/stucky/catkin_ws/src/darknet53 -s /home/stucky/catkin_ws/src/rangenet_lib/example/000000.bin --verbose
> ==18406== Memcheck, a memory error detector
> ==18406== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
> ==18406== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
> ==18406== Command: ./devel/lib/rangenet_lib/infer -p /home/stucky/catkin_ws/src/darknet53 -s /home/stucky/catkin_ws/src/rangenet_lib/example/000000.bin --verbose
> ==18406== 
> --18406-- Valgrind options:
> --18406--    -v
> --18406-- Contents of /proc/version:
> --18406--   Linux version 4.15.0-76-generic (buildd@lgw01-amd64-023) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)) #86~16.04.1-Ubuntu SMP Mon Jan 20 11:02:50 UTC 2020
> --18406-- 
> --18406-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-avx-avx2-bmi
> --18406-- Page sizes: currently 4096, max supported 4096
> --18406-- Valgrind library directory: /usr/lib/valgrind
> --18406-- Reading syms from /home/stucky/catkin_ws/devel/.private/rangenet_lib/lib/rangenet_lib/infer
> --18406-- Reading syms from /lib/x86_64-linux-gnu/ld-2.23.so
> --18406--   Considering /lib/x86_64-linux-gnu/ld-2.23.so ..
> --18406--   .. CRC mismatch (computed 10d8ea02 wanted 4fdda1aa)
> --18406--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/ld-2.23.so ..
> --18406--   .. CRC is valid
> --18406-- Reading syms from /usr/lib/valgrind/memcheck-amd64-linux
> --18406--   Considering /usr/lib/valgrind/memcheck-amd64-linux ..
> --18406--   .. CRC mismatch (computed eea41ea9 wanted 2009db78)
> --18406--    object doesn't have a symbol table
> --18406--    object doesn't have a dynamic symbol table
> --18406-- Scheduler: using generic scheduler lock implementation.
> --18406-- Reading suppressions file: /usr/lib/valgrind/default.supp
> ==18406== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-18406-by-stucky-on-???
> ==18406== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-18406-by-stucky-on-???
> ==18406== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-18406-by-stucky-on-???
> ==18406== 
> ==18406== TO CONTROL THIS PROCESS USING vgdb (which you probably
> ==18406== don't want to do, unless you know exactly what you're doing,
> ==18406== or are doing some strange experiment):
> ==18406==   /usr/lib/valgrind/../../bin/vgdb --pid=18406 ...command...
> ==18406== 
> ==18406== TO DEBUG THIS PROCESS USING GDB: start GDB like this
> ==18406==   /path/to/gdb ./devel/lib/rangenet_lib/infer
> ==18406== and then give GDB the following command
> ==18406==   target remote | /usr/lib/valgrind/../../bin/vgdb --pid=18406
> ==18406== --pid is optional if only one valgrind process is running
> ==18406== 
> --18406-- REDIR: 0x401cf10 (ld-linux-x86-64.so.2:strlen) redirected to 0x3809e181 (???)
> --18406-- Reading syms from /usr/lib/valgrind/vgpreload_core-amd64-linux.so
> --18406--   Considering /usr/lib/valgrind/vgpreload_core-amd64-linux.so ..
> --18406--   .. CRC mismatch (computed 2567ccf6 wanted 49420590)
> --18406--    object doesn't have a symbol table
> --18406-- Reading syms from /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so
> --18406--   Considering /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so ..
> --18406--   .. CRC mismatch (computed 0e27c9a8 wanted ac585421)
> --18406--    object doesn't have a symbol table
> ==18406== WARNING: new redirection conflicts with existing -- ignoring it
> --18406--     old: 0x0401cf10 (strlen              ) R-> (0000.0) 0x3809e181 ???
> --18406--     new: 0x0401cf10 (strlen              ) R-> (2007.0) 0x04c31020 strlen
> --18406-- REDIR: 0x401b860 (ld-linux-x86-64.so.2:index) redirected to 0x4c30bc0 (index)
> --18406-- REDIR: 0x401ba80 (ld-linux-x86-64.so.2:strcmp) redirected to 0x4c320d0 (strcmp)
> --18406-- REDIR: 0x401dc70 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x4c35270 (mempcpy)
> --18406-- Reading syms from /home/stucky/catkin_ws/devel/.private/rangenet_lib/lib/librangenet_lib.so
> --18406-- Reading syms from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_viz3.so.3.3.1
> --18406--    object doesn't have a symbol table
> --18406-- Reading syms from /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.58.0
> --18406--    object doesn't have a symbol table
> --18406-- Reading syms from /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0
> --18406--    object doesn't have a symbol table
> --18406-- Reading syms from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3.1
> --18406--    object doesn't have a symbol table
[... 276 lines cut ...]
> ==18406== Searching for pointers to 37,856 not-freed blocks
> ==18406== Checked 10,693,528 bytes
> ==18406== 
> ==18406== LEAK SUMMARY:
> ==18406==    definitely lost: 0 bytes in 0 blocks
> ==18406==    indirectly lost: 0 bytes in 0 blocks
> ==18406==      possibly lost: 307,455 bytes in 2 blocks
> ==18406==    still reachable: 3,182,097 bytes in 37,854 blocks
> ==18406==                       of which reachable via heuristic:
> ==18406==                         stdstring          : 76,664 bytes in 1,011 blocks
> ==18406==         suppressed: 0 bytes in 0 blocks
[... cut ...]
> ==18406==    by 0x4E5EC78: rangenet::segmentation::NetTensorRT::NetTensorRT(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (netTensorRT.cpp:53)
> ==18406==    by 0x4E5BE16: rangenet::segmentation::make_net(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (selector.cpp:46)
> ==18406==    by 0x404F2C: main (infer.cpp:93)
> ==18406==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
> ==18406== 
> ==18406== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
> Segmentation fault (core dumped)

Spartacus782, 2020-02-04:
I think I have resolved all the issues.  Still not sure exactly what the issue was but it was install related.  Reinstalled the cuda/tensorRT stack again and it worked this time.
Thanks for the help!

closed 2020-02-04 by Spartacus782

ojoo432, 2020-02-06:
Hi @Spartacus782 ~ I am having the same problem now.  Which Cuda、TensorRT、Ubuntu  version are you using,  I am using cuda 10.0 (local deb), TensorRT 5.1.5.0 GA for Ubuntu 1604 and CUDA 10.0 DEB local repo packages. But there is still the problem.

Spartacus782, 2020-02-07:
I think cuda 10.1 is the prereq for TensorRT 5.1.  I also used RC, not sure if that matters.  I started from scratch ripping everything out down to the nvidia drivers and following the install directions super closely.  I also avoided installing anything that wasn't strictly necessary.

> Hi @Spartacus782 ~ I am having the same problem now. Which Cuda、TensorRT、Ubuntu version are you using, I am using cuda 10.0 (local deb), TensorRT 5.1.5.0 GA for Ubuntu 1604 and CUDA 10.0 DEB local repo packages. But there is still the problem.