PRBonn/rangenet_lib #23: Point cloud reflectivity not within 0 to 1, segmentation fail to generate a reasonable data

carlin314, 2020-02-13:
Kitti dataset reflectance within 0 to 1, 

but other dataset may be 0 to 256,

I test dataset(hdl64 dataset from autoware rosbag store) like that(reflectivity within 0 - 255) and find the result was not reasonable,

I dont really understand that and how can I convert the intensity to kitti's?

It seems not divide by 256, or scale by 0.01,

or some more complex regularization like calculate avg,std, do (x-avg)/std,

all these generate bad results?

Do I miss something important?

Or some parameters shoud be change?

Or maybe retrain?

Hope you guys can give me some advices.

carlin314, 2020-02-13:
I dont understand why relfectivity matters here, I thought only geometry information (generated range image) are involved in computation

tano297, 2020-02-13:
As it's said in the [original paper](http://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/milioto2019iros.pdf), the 5 channel input is range, x, y, z, and remission (or as you are calling it, reflectivity).
Which dataset are you using? Which sensor is it?

carlin314, 2020-02-13:
> As it's said in the [original paper](http://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/milioto2019iros.pdf), the 5 channel input is range, x, y, z, and remission (or as you are calling it, reflectivity).
> Which dataset are you using? Which sensor is it?

Thx for replying, I use dataset from autoware rosbag store(a japan website which can download rosbag data)

The sensor is: Velodyne HDL64-s2, 

the data's reflectivity is within 0 - 255,

Do you have any ideas why kitti reflectance is within 0-1? 

Recent data are mostly in 0 - 255

tano297, 2020-02-13:
It's probably just a division by 255. The proper normalization afterwards is already taken care of by [this line](https://github.com/PRBonn/rangenet_lib/blob/f1114af26cd78dffd1d08733e9ba51368154404e/src/netTensorRT.cpp#L250)

carlin314, 2020-02-13:
> As it's said in the [original paper](http://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/milioto2019iros.pdf), the 5 channel input is range, x, y, z, and remission (or as you are calling it, reflectivity).
> Which dataset are you using? Which sensor is it?

rosbag store: https://data.tier4.jp/index/

[image]

carlin314, 2020-02-13:
> It's probably just a division by 255. The proper normalization afterwards is already taken care of by [this line](https://github.com/PRBonn/rangenet_lib/blob/f1114af26cd78dffd1d08733e9ba51368154404e/src/netTensorRT.cpp#L250)

Thx for replying, I will retry it

carlin314, 2020-02-13:
> It's probably just a division by 255. The proper normalization afterwards is already taken care of by [this line](https://github.com/PRBonn/rangenet_lib/blob/f1114af26cd78dffd1d08733e9ba51368154404e/src/netTensorRT.cpp#L250)

Hi, I think maybe the following changed will output somthing better, so I wonder if the following is some statistics over a  whole dataset or just a single cloud?
```
    img_means: #range,x,y,z,signal
      - 12.12
      - 10.88
      - 0.23
      - -1.04
      - 0.21
    img_stds: #range,x,y,z,signal
      - 12.32
      - 11.47
      - 6.91
      - 0.86
      - 0.16
```

Chen-Xieyuanli, 2020-03-08:
As far as I know, the hyperparameter is determined using the validation set, as can be found in the original paper: "We used sequence 08 as validation set for hyperparameter selection and trained our approach on the remaining training sequences. "

@tano297 may give you more accurate information. You could also find more information in the original repo of rangenet++: https://github.com/PRBonn/lidar-bonnetal.

I hope this will help.

Chen-Xieyuanli, 2020-04-01:
Since there is no further update, I'll close this issue. Please feel free to ask me to reopen it if needed.

closed 2020-04-01 by Chen-Xieyuanli