PRBonn/rangenet_lib #24: how to process LIDAR intensity

Oasis-Guo, 2020-02-27:
Hi, thanks for the research result. I manage to implement it at my self-driving, but the result did not meet my expectations.  I think the reason is:

1: I am using 16 lidar from other vendor, not the 64 Velodyne Lidar

2: I am confused with the intensity value. Normally, it should be in 0~255. But I found that the 000000.bin example  lies within <0.5 mostly.  Could U explain a liite bit about it. I could not found the detail online.

Chen-Xieyuanli, 2020-03-03:
Hey @Oasis-Guo,

Sorry for the late reply.

1. Yes, currently the rangenet-lib only works with the KITTI dataset. For more possible applications of rangenet++ you may have a look at [LiDAR-bonnetal](https://github.com/PRBonn/lidar-bonnetal).

2. For intensity value, we follow the typical setup of the KITTI dataset. You may have a look at issue #23. Andres provided more detailed information about LiDAR intensity there.

Hope this helps!

Chen-Xieyuanli, 2020-03-08:
Since this issue is quite similar to #23, I'm going to close this one.

Please feel free to ask me to reopen this one if needed.

closed 2020-03-08 by Chen-Xieyuanli