PRBonn/rangenet_lib #26: ERROR: could not create engine from ONNX

robotichustle, 2020-04-05:
Thanks for your work. I am really excited to demo it but I'm unable to run the demo. 

I am using Ubuntu 18.04 + Cuda 10.1 + CuDNN 7.5.1 + TensorRT 5.1.5 (as per your README and compatibility of versions from Nvidia). GPU is GeForce GTX 1060. 

I get the following runtime error as it can't generate the TRT file from the SemanticKitti darknet53 model:

```
(rangelib_venv) robot@robot-01:~/catkin_ws$ ./devel/lib/rangenet_lib/infer -p ~/datasets/semantic_kitti/models/darknet53 -s ./src/rangenet_lib/example/000000.bin --verbose
================================================================================
scan: ./src/rangenet_lib/example/000000.bin
path: /home/robot/datasets/semantic_kitti/models/darknet53/
verbose: 1
================================================================================
Setting verbosity to: false
Trying to open model
Trying to deserialize previously stored: /home/robot/datasets/semantic_kitti/models/darknet53//model.trt
Could not deserialize TensorRT engine.
Generating from sratch... This may take a while...
Trying to generate trt engine from : /home/robot/datasets/semantic_kitti/models/darknet53//model.onnx
Platform DOESN'T HAVE fp16 support.
No DLA selected.
Could not open file /home/robot/datasets/semantic_kitti/models/darknet53//model.onnx
Could not open file /home/robot/datasets/semantic_kitti/models/darknet53//model.onnx
Failed to parse ONNX model from file/home/robot/datasets/semantic_kitti/models/darknet53//model.onnx
Success picking up ONNX model
Failure creating engine from ONNX model
Current trial size is 8589934592
Failure creating engine from ONNX model
Current trial size is 4294967296
Failure creating engine from ONNX model
Current trial size is 2147483648
Failure creating engine from ONNX model
Current trial size is 1073741824
Failure creating engine from ONNX model
Current trial size is 536870912
Failure creating engine from ONNX model
Current trial size is 268435456
Failure creating engine from ONNX model
Current trial size is 134217728
Failure creating engine from ONNX model
Current trial size is 67108864
Failure creating engine from ONNX model
Current trial size is 33554432
Failure creating engine from ONNX model
Current trial size is 16777216
Failure creating engine from ONNX model
Current trial size is 8388608
Failure creating engine from ONNX model
Current trial size is 4194304
Failure creating engine from ONNX model
Current trial size is 2097152
Failure creating engine from ONNX model
Current trial size is 1048576
terminate called after throwing an instance of 'std::runtime_error'
  what():  ERROR: could not create engine from ONNX.
Aborted (core dumped)
```

I am not sure if it matters but the catkin build process passed but I got some warnings. Here is the output:

```
(rangelib_venv) robot@robot-01:~/catkin_ws$ catkin build rangenet_lib
----------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/robot/catkin_ws
----------------------------------------------------------
Build Space:        [exists] /home/robot/catkin_ws/build
Devel Space:        [exists] /home/robot/catkin_ws/devel
Install Space:      [unused] /home/robot/catkin_ws/install
[... 102 lines cut ...]

Building TensorRT
Building example...
OpenCV Libs: opencv_core;opencv_viz
OpenCV Headers: /usr/include;/usr/include/opencv

cd /home/robot/catkin_ws/build/rangenet_lib; catkin build --get-env rangenet_lib | catkin env -si  /usr/local/bin/cmake /home/robot/catkin_ws/src/rangenet_lib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/robot/catkin_ws/devel/.private/rangenet_lib -DCMAKE_INSTALL_PREFIX=/home/robot/catkin_ws/install; cd -
............................................................................................................................................................................................................................................
Finished  <<< rangenet_lib                [ 8.6 seconds ]
[build] Summary: All 2 packages succeeded!
[build]   Ignored:   None.
[build]   Warnings:  1 packages succeeded with warnings.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 8.7 seconds total.
Exception ignored in: <bound method BaseEventLoop.__del__ of <_UnixSelectorEventLoop running=False closed=True debug=False>>
Traceback (most recent call last):
  File "/home/robot/.virtualenvs/rangelib_venv/lib/python3.6/site-packages/trollius/base_events.py", line 395, in __del__
  File "/home/robot/.virtualenvs/rangelib_venv/lib/python3.6/site-packages/trollius/unix_events.py", line 65, in close
  File "/home/robot/.virtualenvs/rangelib_venv/lib/python3.6/site-packages/trollius/unix_events.py", line 166, in remove_signal_handler
  File "/usr/lib/python3.6/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object
```

Any help will be much appreciated.

Chen-Xieyuanli, 2020-04-06:
Hey @robotichustle,

Thank you for using our code.

The catkin build TypeError is a Python-related issue, see more [here](https://bugs.python.org/issue26133). I'm not sure whether this is related to the TRT model problem or not.

For the TRT model problem, it should be a TensorRT-related issue. We never encountered this error by ourselves and it's hard for us to reproduce it and debug it.

However, there are some similar issues reported by others. You may have a look there #15 #22.

Chen-Xieyuanli, 2020-04-28:
Hey @robotichustle, is there any update on this issue?

ZhenminHuang, 2020-05-07:
@robotichustle 
Previously I was using CUDA10.1+cudnn7.6.5+trt5.1.5 with NVIDIA P1000, and I ran into exactly the same problem. I tried downgrading CUDA10.1 to CUDA10.0 and it works well now.
I also tested this workaround with another device with GTX 1060, and it works well too.
I suggest maybe trt5.1.5 is somehow incompatible with CUDA version higher than 10.0?

Chen-Xieyuanli, 2020-05-19:
@ZhenminHuang Thank you so much for the feedback!

Since there is no further update about this issue, I'm going to close it. Please feel free to ask me to reopen it if needed.

kuzen, 2020-06-09:
check you cublas version == cuda version

GuoFeng-X, 2020-12-12:
I also tested with GTX1660 + cuda10.0 + cudnn7.5.0 + TensorRT5.1.5.0, and it works well.

closed 2021-06-07 by Chen-Xieyuanli

xdtzzz, 2022-01-12:
hi, @GuoFeng-X  ,  你的显卡内存是多少呢

caobugai12138, 2022-10-29:
I have the same problem， is anybody can propose a solution？