PRBonn/rangenet_lib #28: Performance on other data?

kxhit, 2020-05-15:
Hi guys! Have anyone tried the pretrained model on other datasets or in realistic environment? I'm curious the transfer ability and how to preprocess data for better results. Welcome to share your experience! Thanks!

Chen-Xieyuanli, 2020-05-17:
Hey @kxhit, I tried the pre-trained model with the Ford Campus dataset and it didn't work very well. I guess the reason is that the environment of the Ford Campus dataset is quite different from that of the KITTI dataset and the pre-trained cannot generalize well in such different environments.

If you want the pre-trained model to work well in other environments, you may need to fine-tune it with data from the new environments.

kxhit, 2020-05-17:
Hi @Chen-Xieyuanli  I'm trying to test it on Ford Campus dataset right now!
I'm preprocessing the dataset to get the intensity value by running 'create_ijrr_dataset.m'.  When I tried on IJRR-Dataset-1, the size of SCAN.XYZ is different with SCAN.reflectivity, which should be the same. However, when I tried on IJRR-Dataset-2, they have the same shape, which is right! Have you ever had such a problem? Thanks if you could help me!

I will give the test performance once I have the result.

Chen-Xieyuanli, 2020-05-19:
You may check both the SCANS files and the VELODYNE files. The reflectivities are contained in the VELODYNE files. For more details about preprocessing the Ford Campus dataset, please find it in their provided [code](http://robots.engin.umich.edu/uploads/SoftwareData/Ford/Code.zip).

kxhit, 2020-05-19:
@Chen-Xieyuanli Sorry to interrupt. 
I solved this problem by adding some code in 'create_ijrr_dataset.m'. The reason is that IJRR-Dataset-1 is begin with Scan0075.mat, so the 'start_index' should be modified.

Adding code from line 37 in 'create_ijrr_dataset.m'

```
    if numFiles == 3    % dataset-2
        start_index = 1; %1
    else                % dataset-1
        if fileIndex == 0
            start_index = 75; % begin with Scan0075
        else
            start_index = 1;
        end
    end
```

Chen-Xieyuanli, 2020-05-19:
As far as I know, there is already an "offset" variable in their code which you could use to align different data.

Anyway, glad to hear that the problem is solved!!

kxhit, 2020-06-10:
I got predictions like this.  Emmm, it seems that generalization ability is still a thorny problem to solve.

[image]

closed 2020-06-10 by kxhit

Chen-Xieyuanli, 2020-06-10:
@kxhit, Thank you for your feedback!

GuoFeng-X, 2020-12-21:
> Hi guys! Have anyone tried the pretrained model on other datasets or in realistic environment? I'm curious the transfer ability and how to preprocess data for better results. Welcome to share your experience! Thanks!

Hi! Have anyone tried the pretrained model  in realistic environment? such as VLP-32 or VLP-16?  Which parameters do you need to modify？Looking forward to your reply， thank you!

xdtzzz, 2021-11-16:
hi,@GuoFeng-X, have you tried other dataset in realistic environment? Did you succeed?

xdtzzz, 2021-11-16:
Hi,@kxhit, Do you have any new progress in training other dataset？

G12311231, 2024-07-02:
> 嘿@kxhit，我尝试使用福特校园数据集来训练预训练模型，但效果不太好。我猜原因是福特校园数据集的环境与 KITTI 数据集的环境完全不同，而预训练模型无法在如此不同的环境中很好地推广。
> 
> 如果您希望预训练模型在其他环境中运行良好，则可能需要使用来自新环境的数据对其进行微调。
Hello, 
Does the Ford data set also use vlp-64 to collect data?