PRBonn/rangenet_lib #29: Catkin build error

kxhit, 2020-05-15:
Hello~ I met an error when build the lib. Could you help me? Thanks! 
My environment is Ubuntu16.04 python3.5 TensorRT5.1 CUDA10.0.

```
 catkin build rangenet_lib
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/kx/3rdlib/TensorRT-5.1.2.2/lib
Workspace:                   /media/work/3D/rangenet_lib/catkin_ws
Source Space:       [exists] /media/work/3D/rangenet_lib/catkin_ws/src
Build Space:        [exists] /media/work/3D/rangenet_lib/catkin_ws/build
Devel Space:        [exists] /media/work/3D/rangenet_lib/catkin_ws/devel
Install Space:     [missing] /media/work/3D/rangenet_lib/catkin_ws/install
DESTDIR:                     None
--------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------- 
Found '2' packages in 0.0 seconds. 
Starting ==> catkin                                                                                                                                
Finished <== catkin       [ 1.6 seconds ]                                                                                                          
Starting ==> rangenet_lib                                                                                                                          
                                                                                                                                                   
[rangenet_lib] ==> '/media/work/3D/rangenet_lib/catkin_ws/build/rangenet_lib/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/media/work/3D/rangenet_lib/catkin_ws/build/rangenet_lib'
[ 25%] Built target rangenet_external
[ 37%] Linking CXX shared library /media/work/3D/rangenet_lib/catkin_ws/devel/lib/librangenet_lib.so
/usr/bin/ld:  -lnvinfer
/usr/bin/ld:  -lnvinfer_plugin
/usr/bin/ld:  -lnvparsers
/usr/bin/ld:  -lnvonnxparser
/usr/bin/ld:  -lnvonnxparser_runtime
collect2: error: ld returned 1 exit status
CMakeFiles/rangenet_lib.dir/build.make:150: recipe for target '/media/work/3D/rangenet_lib/catkin_ws/devel/lib/librangenet_lib.so' failed
make[2]: *** [/media/work/3D/rangenet_lib/catkin_ws/devel/lib/librangenet_lib.so] Error 1
CMakeFiles/Makefile2:334: recipe for target 'CMakeFiles/rangenet_lib.dir/all' failed
make[1]: *** [CMakeFiles/rangenet_lib.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
[rangenet_lib] <== '/media/work/3D/rangenet_lib/catkin_ws/build/rangenet_lib/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
                                                                                                                                                    
Failed   <== rangenet_lib [ 0.3 seconds ]                                                                                                          
[build] There were '1' errors:                                                                                                                     

Failed to build package 'rangenet_lib' because the following command:

# Command to reproduce:
cd /media/work/3D/rangenet_lib/catkin_ws/build/rangenet_lib && /media/work/3D/rangenet_lib/catkin_ws/build/rangenet_lib/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /media/work/3D/rangenet_lib/catkin_ws/build/build_logs/rangenet_lib.log

Exited with return code: 2 
```

Chen-Xieyuanli, 2020-05-17:
Hey @kxhit, this seems to be a linking problem of the tensorrt library. You may first have a look at a similar issue here #18.

kxhit, 2020-05-17:
Thanks a lot! It goes well now.

closed 2020-05-17 by kxhit