PRBonn/rangenet_lib #31: How to save the result label data

CK-Sung, 2020-08-07:
Hi, I am running the rangenet++ and also the OverlapNet.

The OverlapNet can be run with depth, normal, and intensity. But, I would like to add the label data as well.

So, I tested the RangeNet++ to generate the label data with pre-trained weight, but the code only shows the visualization results.

How can I save the label data which is the result of the RangeNet++ to use in the OverlapNet?

Chen-Xieyuanli, 2020-08-07:
Hey @CK-Sung, you could have a look at the [infer](https://github.com/PRBonn/rangenet_lib/blob/master/src/netTensorRT.cpp) function, which returns the semantic estimates with probabilities over all classes with the shape of (_n_classes, _img_h, _img_w).

CK-Sung, 2020-08-07:
Thank you so much !

Chen-Xieyuanli, 2020-08-07:
@CK-Sung Thank you for following our work, and I will appreciate it a lot if you could star or fork our repo too.

CK-Sung, 2020-08-07:
@Chen-Xieyuanli Hi, now I know what is the result of the semantic_scan. But, I am still not sure the data type of the label file.

CK-Sung, 2020-08-07:
Hi @Chen-Xieyuanli , I got another question about the .label file.

Now I am comparing the data between the 'semantic_scan result with sequence/00/000000.bin file' and 000000.label file in OverlapNet/data/semantic_probs.

Here are two questions

1. Is the 000000.label file in the OverlapNet/data/semantic_probs result of the sequnce00/000000.bin file?
- I have compared values. First vector of the semantic_scan: "3.22323e-09, 2.36382e-06, ...., 0.995063, .., 8.71254e-06"
  label data: "bab8 e930 263e 9b35 e6ac 3431 0e6a 9832 ..... 30c8 7e3f abb2 c737 539b 3238 a829 3137"
2. If YES for the question 1, how can I change the probability values to save as correct data type in .label file?

Thank you

closed 2020-08-09 by CK-Sung

BlingLee, 2024-09-05:
> Thank you so much !
Hi, I'm a beginner and I wanted to ask, How can I save the label data which is the result of the RangeNet++ to use in the OverlapNet?
“you could have a look at the infer function,  which returns the semantic estimates with probabilities over all classes with the shape of (_n_classes, _img_h,  What does "_img_w)" mean?   I can't generate semantic data .Could you give me some advice? Looking for your reply!