PRBonn/rangenet_lib #32: Hardcoded values for the FoV on spherical projection

kosmastsk, 2020-11-11:
In [netTensorRT.cpp](https://github.com/PRBonn/rangenet_lib/blob/master/src/netTensorRT.cpp#L123) file in the function `doProjection()` the values of the sensor's Field of View are hardcoded as: -25 to +3 degrees.

In case that a different sensor is used with a different FoV, shouldn't these values be grabbed from the `arch_cfg.yaml` file that is in the (pre-trained or not) model folder and consequently in the model.onnx that has been created?

Chen-Xieyuanli, 2020-11-17:
Hey @kosmastsk,

Thank you for your suggestion! I may update it later.

kosmastsk, 2020-11-17:
@Chen-Xieyuanli thank you for your response. If you provide me with some guidelines, I could make this change and submit a PR.

Chen-Xieyuanli, 2020-11-19:
@kosmastsk That would be great! You could find the parameters loaded in net.cpp and may add two more global variables there as fov_up and fov_down.

kosmastsk, 2020-11-19:
Alright @Chen-Xieyuanli . I'll take a look and probably I'll fix this soon!

PR #34 (2021-09-07): Remove hardcoded FoV values in spherical projection
Fixed #32
closed 2021-09-07 by Chen-Xieyuanli