PRBonn/rangenet_lib #33: can i use Cmake not catkin_make to build it?

shqfeng, 2020-11-13:


Chen-Xieyuanli, 2020-11-17:
Hey @shqfeng,

I never tried that before, but I think you could achieve that with some small modification.

We previously used the catkin build to enable the ROS wrapper. Since the ROS wrapper is not included anymore, it should be possible to also get rid of the catkin build.

shqfeng, 2020-11-17:
thanks your reply，i have made it sucessfully in windows------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;Xieyuanli&amp;nbsp;Chen&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年11月17日(星期二) 下午5:43
收件人:&nbsp;&quot;PRBonn/rangenet_lib&quot;<rangenet_lib@noreply.github.com&gt;;
抄送:&nbsp;&quot;Shq feng&quot;<602414626@qq.com&gt;;&quot;Mention&quot;<mention@noreply.github.com&gt;;
主题:&nbsp;Re: [PRBonn/rangenet_lib] can i use Cmake not catkin_make to build it? (#33)

Chen-Xieyuanli, 2020-11-17:
That's great! I'm glad it works for you!

I will appreciate it a lot if you could provide more information here for other users.

Since it works now, I will close this issue.

closed 2020-11-17 by Chen-Xieyuanli