PRBonn/rangenet_lib #37: SyntaxError: invalid syntax When build library

xhding1997, 2021-03-05:
I have followed these steps and stuck at  "cd .. && catkin init".
$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ git clone https://github.com/ros/catkin.git 
$ git clone https://github.com/PRBonn/rangenet_lib.git
$ cd .. && catkin init
$ catkin build rangenet_lib

bie@bie:~/catkin_ws$ catkin init
Traceback (most recent call last):
  File "/usr/local/bin/catkin", line 5, in <module>
    from catkin_tools.commands.catkin import main
  File "/usr/local/lib/python3.8/dist-packages/catkin_tools/commands/catkin.py", line 28, in <module>
    from catkin_tools.common import is_tty
  File "/usr/local/lib/python3.8/dist-packages/catkin_tools/common.py", line 23, in <module>
    import trollius as asyncio
  File "/usr/local/lib/python3.8/dist-packages/trollius/__init__.py", line 21, in <module>
    from .base_events import *
  File "/usr/local/lib/python3.8/dist-packages/trollius/base_events.py", line 42, in <module>
    from . import tasks
  File "/usr/local/lib/python3.8/dist-packages/trollius/tasks.py", line 565
    def async(coro_or_future, loop=None):
        ^
SyntaxError: invalid syntax

How to solve this ?Is it a python version problem? But I have tried python2.7 and 3.8 both, it doesn't work.

jbehley, 2021-03-05:
the version of catkin_tools seems to be incompatible with python 3.8+, see https://github.com/catkin/catkin_tools/issues/594#issuecomment-635621597  . Either get an updated catkin_tools or use the workaround suggested in the linked vcomment.

xhding1997, 2021-03-06:
> the version of catkin_tools seems to be incompatible with python 3.8+, see [catkin/catkin_tools#594 (comment)](https://github.com/catkin/catkin_tools/issues/594#issuecomment-635621597) . Either get an updated catkin_tools or use the workaround suggested in the linked vcomment.

Thanks for your answer! It really works when I use this :
sudo apt install python3-catkin-tools python3-osrf-pycommon

But I got a new trouble here. TensorRT doesn't support Ubuntu20.04 right now, do you have any suggestion for this?

pjs9115916, 2021-09-26:
`catkin init
Traceback (most recent call last):
  File "/usr/local/bin/catkin", line 9, in <module>
    load_entry_point('catkin-tools==0.7.1', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 200
    file=sys.stderr)
        ^
SyntaxError: invalid syntax`

when I input catkin init , a error occur, how to solve it?
whether I can use catkin_make to replace catkin init and catkin build rangenet_lib?
[image]

pjs9115916, 2021-09-27:
> Hey @pjs9115916, thanks for using our code. As discussed above, this is an incompatible problem between catkin_tools and python. You may find more details in catkin_tool repo here. We so far have no solution for this issue and warmly welcome you to report your discovery here later.

Thank you for your reply, actually ,I have builded passed through catkin_make. but I get a bad result, I am not sure whether it is because of my build method. can you figure out it for me ?

this is my bad result.

pjs9115916, 2021-09-27:
Thank you for your reply, actually ,I have builded passed through catkin_make. but I get a bad result, I am not sure whether it is because of my build method. can you figure out it for me ?

this is my bad result.
[image]

Chen-Xieyuanli, 2021-09-28:
Hey @pjs9115916, this issue has been well discussed in #9. You may find the solution there.

It would be very nice of you if you could report here how you solve the SyntaxError which other users will benefit from.

pjs9115916, 2021-09-28:
OK，thanks a lot.I don't follow your step.Just follow the step of ros catkin_make, and then build passed.发自我的荣耀手机-------- 原始邮件 --------发件人： Xieyuanli Chen ***@***.***>日期： 2021年9月28日周二 傍晚6:37收件人： PRBonn/rangenet_lib ***@***.***>抄送： pjs9115916 ***@***.***>, Mention ***@***.***>主    题： Re: [PRBonn/rangenet_lib] SyntaxError: invalid syntax When build library (#37)
Hey @pjs9115916, this issue has been well discussed in #9. You may find the solution there.
It would be very nice of you if you could report here how you solve the SyntaxError which other users will benefit from.

Chen-Xieyuanli, 2021-09-28:
Thanks for sharing your solution. Since this issue was solved, I would like to close it.

closed 2021-09-28 by Chen-Xieyuanli

lifangyan16, 2022-01-19:
> 我遵循了这些步骤并卡在"cd..&& catkin init"。$ mkdir -p ~/catkin_ws/src$ cd ~/catkin_ws/src$ git clone https://github.com/ros/catkin.git$ git clone https://github.com/PRBonn/rangenet_lib.git$ 光盘 ..&& catkin init$ 卡特金构建rangenet_lib
> 
> bie@bie：~/catkin_ws美元回溯（最近一次调用最后一次）：文件"/usr/local/bin/catkin"，第 5 行，在from catkin_tools.commands.catkin import main文件"/usr/local/lib/python3.8/dist-packages/catkin_tools/commands/catkin.py"，第 28 行，在从 catkin_tools.common import is_tty文件"/usr/local/lib/python3.8/dist-packages/catkin_tools/common.py"，第 23 行，在import trollius as asyncio文件 "/usr/local/lib/python3.8/dist-packages/trollius/**init**.py"，第 21 行，在从.base_events导入 *文件"/usr/local/lib/python3.8/dist-packages/trollius/base_events.py"，第 42 行，在从。导入任务文件"/usr/local/lib/python3.8/dist-packages/trollius/tasks.py"，第 565 行def async（coro_or_future， loop=None）： ^语法错误：语法无效
> 
> 如何解决这个问题？是 Python 版本问题吗？但是我都尝试过python2.7和3.8，它不起作用。

Did you use ubuntu16?Use ubuntu18.