PRBonn/rangenet_lib #38: could not parse input ONNX.

xdtzzz, 2021-06-07:
hi, thx for your amazing codes.
when i trained my own dataset using your (https://github.com/PRBonn/lidar-bonnetal) , i got a file, like this.
[image]

and then i change it to a model.onnx,   then i want to use rangenet_lib to check my train performence, but a problem happens, like this.
[image]
[image]
it seems that there is a problem in model.onnx.
can you help me? thanks in advance!!!

Chen-Xieyuanli, 2021-06-07:
Hi @xdtzzz, this issue is quite similar to #26. Could you please first have a look there?

Since this is duplicated to #26, I would like to close this one to better track the same issue.

xdtzzz, 2021-06-07:
thanks for your reply so quickly.
but my problem is different to that.
that question is test a demo on your trained model, but my question is on my own trained model.
i can test it successfully on your trained model, but cannot work on my own.

closed 2021-11-16 by xdtzzz