PRBonn/rangenet_lib #41: compile error: No module named catkin_pkg.package

zhjw0927, 2021-11-04:
I have build a docker container from nvidia/cuda:1.0-base
After I installed the system and python dependencies in the readme, I execute catkin build rangenet_lib, but the following error occurred:
x-special/nautilus-clipboard
copy
file:///home/zhjw/Pictures/Screenshot%20from%202021-11-04%2001-37-54.png
I verified that the library can be imported correctly, as shown below:
x-special/nautilus-clipboard
copy
file:///home/zhjw/Pictures/Screenshot%20from%202021-11-04%2001-42-38.png
I dont know how to solve, can you help me?

zhjw0927, 2021-11-04:
i try export "catkin_pkg" into PYTHONPATH, new issue occurred:
**Traceback (most recent call last):
  File "/home/zhjw/code/catkin_ws/src/catkin/cmake/parse_package_xml.py", line 40, in <module>
    from catkin_pkg.package import parse_package
  File "/usr/local/lib/python3.6/dist-packages/catkin_pkg/package.py", line 44, in <module>
    from catkin_pkg.condition import evaluate_condition
  File "/usr/local/lib/python3.6/dist-packages/catkin_pkg/condition.py", line 17, in <module>
    import pyparsing as pp
  File "/usr/local/lib/python3.6/dist-packages/pyparsing/__init__.py", line 115, in <module>
    from .util import *
  File "/usr/local/lib/python3.6/dist-packages/pyparsing/util.py", line 16
    _all_names: List[str] = []
              ^
SyntaxError: invalid syntax**
hope to get your help, thanks!

Chen-Xieyuanli, 2021-11-04:
> i try export "catkin_pkg" into PYTHONPATH, new issue occurred: **Traceback (most recent call last): File "/home/zhjw/code/catkin_ws/src/catkin/cmake/parse_package_xml.py", line 40, in from catkin_pkg.package import parse_package File "/usr/local/lib/python3.6/dist-packages/catkin_pkg/package.py", line 44, in from catkin_pkg.condition import evaluate_condition File "/usr/local/lib/python3.6/dist-packages/catkin_pkg/condition.py", line 17, in import pyparsing as pp File "/usr/local/lib/python3.6/dist-packages/pyparsing/**init**.py", line 115, in from .util import * File "/usr/local/lib/python3.6/dist-packages/pyparsing/util.py", line 16 _all_names: List[str] = [] ^ SyntaxError: invalid syntax** hope to get your help, thanks!

It seems like a version conflict between python and catkin_pkg. I never met such an error before. I will leave this issue open and see whether others can provide any help.

closed 2022-02-14 by Chen-Xieyuanli