PRBonn/rangenet_lib #47: ‘Net’ was not declared in this scope

mc171819, 2022-05-19:
Hi, when I tried to make this project, it occurs an error: 'Net' was not declared in this scope, in selector.cpp and selector.hpp??? It means that the nettensorrt is not well import? @Chen-Xieyuanli

mc171819, 2022-05-19:
I can't get into verbosity in netTensorRT.cpp and can't output Setting verbosity to:XXX

mc171819, 2022-05-19:
and when I delete net->verbosity(verbose);, I get the same error in net->infer(XXX)，I still cannot get into it. （I have already have a trt model by python so I don't need to generate it by onnx now）.I think may be 'net' is not correct?

Chen-Xieyuanli, 2022-06-14:
@mc171819 Sorry for the late reply. I never got the same problem before. We have several pull requests which you may give a try where other users add some modifications for using tensorrt7