PRBonn/rangenet_lib #55: error when verbose

ContinueLopo, 2023-07-31:
Dear author, hello。
When I run the demo of rangenet_ lib, the following error occurred. I don't know how to solve it. Could you please help me take a look?
```
./devel/lib/rangenet_lib/infer -p /src/rangenet_lib/model/darknet53 -s /src/rangenet_lib/example/000000.bin --verbose
================================================================================
scan: /src/rangenet_lib/example/000000.bin
path: /src/rangenet_lib/model/darknet53/
verbose: 1
================================================================================
terminate called after throwing an instance of 'std::runtime_error'
  what():  Can't open cfg.yaml from /src/rangenet_lib/model/darknet53//arch_cfg.yaml
已放弃 (核心已转储)
```

small-jiang, 2023-11-24:
这个问题解决了吗朋友？

closed 2023-11-28 by ContinueLopo

small-jiang, 2023-11-29:
朋友你好，我遇到了一样的问腿，方便提示一下你是怎么解决的吗，感激！

small-jiang, 2023-11-29:
@ContinueLopo 朋友你好，我遇到了一样的问腿，方便提示一下你是怎么解决的吗，感激！

boyang9602, 2024-08-24:
@small-jiang 
我在semantic_suma里遇到类似问题，是模型路径错误的问题，你仔细检查一下路径
I met a similar problem in semantic_suma. It was due to the incorrect model path. You need to carefully check the path.