PRBonn/rangenet_lib #58: question about pretrained model?

Zhangjyhhh, 2024-02-23:
could you please tell me the difference between your provide pretrain model and the rangenet++ trained mode?
here is my trained model , the backone file seems different from yours
[image]

G12311231, 2024-07-01:
> 你能告诉我你提供的预训练模型和 rangenet++ 训练模式之间的区别吗？ 这是我训练的模型，backone 文件似乎与你的不同 ![氨基酸](https://private-user-images.githubusercontent.com/118284536/307256574-d7a6a874-f69a-4e5b-a914-7ab49f333fcb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4MDA5MzAsIm5iZiI6MTcxOTgwMDYzMCwicGF0aCI6Ii8xMTgyODQ1MzYvMzA3MjU2NTc0LWQ3YTZhODc0LWY2OWEtNGU1Yi1hOTE0LTdhYjQ5ZjMzM2ZjYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwMVQwMjIzNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MTY5NDAxNDNhNWNiNjhkMjYyMzRjMjVjNzlkMzg3NTRjZWI2MTc2MWZjMGIwMDEyYzk1OWIzZDVmM2NkYWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dlNfHQz275yHBSrmXbkfJ-BRj6HzmrTzmt05B5zB6ro)

Hi, I would like to ask how you get the training model. Is it necessary to run lidar-bonnetal? The tensorrt version required is too old. The tensorrt I installed is 8.5.1.7. I would like to know your solution.Thank you!

Zhangjyhhh, 2024-07-01:
> Hi, I would like to ask how you get the training model. Is it necessary to run lidar-bonnetal? The tensorrt version required is too old. The tensorrt I installed is 8.5.1.7. I would like to know [...]

yes, I get my training model by running lidar-bonnetal. there is a solution for higher tensorrt,  you could reference this link[https://github.com/PRBonn/rangenet_lib/issues/9](url)

G12311231, 2024-07-01:
Thank you very much for your answer! It means that when I want to use a custom dataset to train the network, I need to run lidar-bonnetal successfully. Otherwise, I can directly use the model trained by the author to run, refer to rangenet_lib, right?

Zhangjyhhh, 2024-07-01:
> Thank you very much for your answer! It means that when I want to use a custom dataset to train the network, I need to run lidar-bonnetal successfully. Otherwise, I can directly use the model trained by the author to run, refer to rangenet_lib, right?

right!!!